pointhi/kicad-color-schemes #8: Patch.py does not ignore the kicad_advanced config file

nickoe, 2019-09-23:
As documented on http://docs.kicad-pcb.org/doxygen/namespaceAC__KEYS.html

One can place a file named kicad_advanced to toggle some experimental settings, but running patch.py errors:
```
$ python3 patch.py ~/kicad-color-schemes/blue-green-dark/ ~/.config/kicad/
Scheme does not contain a definition for EESchema, skipped.
Traceback (most recent call last):
  File "patch.py", line 18, in __init__
    key, value = l.split('=', 1)
ValueError: not enough values to unpack (expected 2, got 1)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "patch.py", line 103, in <module>
    pcb_handler = ConfigFile(pcb_config)
  File "patch.py", line 20, in __init__
    raise ValueError("line {}: '{}' is of invalid format in config file.".format(idx, l))
ValueError: line 426: '[ImportGraphics]' is of invalid format in config file.```

nickoe, 2019-09-23:
It looks like I have this section in the pcbnew file:
```
[ImportGraphics]                                                   
BoardLayer=40                                                      
InteractivePlacement=1                                             
LastFile=/foo/bar.svg
PositionUnits=0                                                    
PositionX=0                                                        
PositionY=0                                                        
LineWidth=0.2                                                      
LineWidthUnits=0                                                   
```

pointhi, 2019-11-27:
should be fixed now

closed 2019-11-27 by pointhi