pointhi/kicad-color-schemes #11: whish to merge the code with [skalidindi3/kicad-colors](https://github.com/skalidindi3/kicad-colors)

easyw, 2019-11-27:
whish to merge the code with https://github.com/skalidindi3/kicad-colors

poeschlr, 2020-01-15:
It seems the owner of the other repo is not really interested in merging the two. The reason it with "other projects also have a split up community for colour schemes."

Pun-e, 2020-07-08:
The other repo hasn't had a "real" commit in over 2 years, but does have some really nice schemes like onedark, nord and rebecca.

Is there any chance we could have their open-source schemes rolled into the one repo, and hopefully get this unified theme database pushed into KiCad in time so that all users would have access to them?

ddm9599, 2020-11-27:
If someone wants to give me access to this repository, I have a local commit which merges in the other themes, plus does a bunch of other things - including displaying a list of available themes for the user to interactively pick from and correctly making/saving a backup eeschema.

pointhi, 2020-11-27:
Please create Pull Requests for all those things you want to have merged.

ddm9599, 2020-11-27:
Certainly, but I am unable to push my branch. I'm getting authentication errors which suggests I need additional repository permissions?

pointhi, 2020-11-27:
You have to create a personal fork and push to this fork, not onto my repository.

ddm9599, 2020-11-28:
@pointhi Done: https://github.com/pointhi/kicad-color-schemes/pull/22