pointhi/kicad-color-schemes #14: Monokai Color Scheme

Shackmeister, 2020-01-09:
Hi @pointhi Could you please add this to the list?
The color scheme below is the monokai scheme (https://monokai.pro/) most famously known from Sublime text editor.
Definitions found here: http://www.eclipsecolorthemes.org/?view=theme&id=6093
All credit should go to Forum user kickofighto (OP here: https://forum.kicad.info/t/monokai-clone-theme-for-eeschema/12917), I simply updated this to work with 5.1

Color4DWireEx=rgb(166, 226, 46)
Color4DBusEx=rgb(102, 217, 239)
Color4DConnEx=rgb(166, 226, 46)
Color4DLLabelEx=rgb(248, 248, 242)
Color4DHLabelEx=rgb(255, 231, 146)
Color4DGLabelEx=rgb(249, 38, 114)
Color4DPinNumEx=rgb(249, 38, 114)
Color4DPinNameEx=rgb(102, 217, 239)
Color4DFieldEx=rgb(174, 129, 255)
Color4DReferenceEx=rgb(102, 217, 239)
Color4DValueEx=rgb(102, 217, 239)
Color4DNoteEx=rgb(253, 151, 31)
Color4DBodyEx=rgb(249, 38, 114)
Color4DBodyBgEx=rgb(73, 72, 62)
Color4DNetNameEx=rgb(117, 113, 94)
Color4DPinEx=rgb(249, 38, 114)
Color4DSheetEx=rgb(174, 129, 255)
Color4DSheetFileNameEx=rgb(255, 231, 146)
Color4DSheetNameEx=rgb(102, 217, 239)
Color4DSheetLabelEx=rgb(255, 231, 146)
Color4DNoConnectEx=rgb(102, 217, 239)
Color4DErcWEx=rgb(166, 226, 46)
Color4DErcEEx=rgb(249, 38, 114)
Color4DGridEx=rgb(73, 72, 62)
Color4DBgCanvasEx=rgb(59, 58, 50)
Color4DBrighenedEx=rgb(248, 248, 242)

poeschlr, 2020-01-10:
Damn now i worry about licencing problems instead of loving the nice colours

Shackmeister, 2020-01-10:
Quote from monokai author here: https://hackernoon.com/monokai-pro-beautiful-functionality-for-professional-developers-a6b9b283f0b3

"Almost every code editor has a Monokai preset. It has been recycled, remixed and implemented in more software products than I can count. The old Monokai has been going strong for more than a decade."

From what I can tell, plenty OSS tools has the "old" monokai theme. I believe (but do not know) only the new monokai with icons is licensed

closed 2020-03-18 by pointhi