pointhi/kicad-color-schemes #27: Great work - a light background for PCB is also needed.

moderndevice, 2021-07-29:
I realize that a light background for PCBs is problematic at the current time because of the color algorithm which just lightens selected colors. I am using my own light theme which does work - although selected colors are sometimes sub-optimal.

I'll be happy to work on some pleasing colors for a light theme. The algorithm question is another matter though as I think it would require recompiling Kicad to test. Something I'm not up for right now.

pointhi, 2021-07-31:
Hi, you can create a light theme for here and create an issue on the KiCad bugtracker to improve handling of light themes.

moderndevice, 2021-07-31:
Are PCB themes separate from Schema themes? Seems like all the work here is Schema themes?