pointhi/kicad-color-schemes #28: KiCad Plugin and Content Manager addition

pointhi, 2021-10-30:
Dear color theme contributors @poeschlr @esden @DX-MON @craftyjon @wykys 

I was asked to add those color schemes into the Plugin and Content Manager repository. Thus I would like to ask if you are OK with adding you as author (please give me a name, and as contact info I will simply add your github user if you like). As license, I set this repo to CC0-1.0, but I can set a different license on your theme if you like.

Either you create a PR with the given metadata.json file, or I will create one if you are ok with it. The file looks like this and is added to the color-scheme-directory. The directory further has to include the color-scheme in json format which I can then bundle into a package.

```json
{
    "$schema": "https://gitlab.com/qu1ck/kicad/-/raw/pcm/kicad/pcm/schemas/pcm.v1.schema.json#",
    "name": "solarized-light theme",
    "description": "solarized-light color theme",
    "description_full": "solarized-light color theme",
    "identifier": "theme-solarized-light",
    "type": "colortheme",
    "author": {
        "name": "Thomas Pointhuber",
        "contact": {
            "web": "https://github.com/pointhi/kicad-color-schemes/"
        }
    },
    "maintainer": {
        "name": "Thomas Pointhuber",
        "contact": {
            "web": "https://github.com/pointhi/kicad-color-schemes/"
        }
    },
    "license": "CC0-1.0",
    "resources": {
        "Github": "https://github.com/pointhi/kicad-color-schemes"
    },
    "versions": [{
        "version": "1.0",
        "status": "stable",
        "kicad_version": "5.99"
    }]
}
```

craftyjon, 2021-10-30:
The two themes I added to this repo (kicad-classic and kicad-default) should not be included in the PCM as they are built-in to KiCad.

DX-MON, 2021-10-30:
I am happy for the Eagle-style theme to be included.
Please use ***@***.*** as contact info, and Rachel Mant as the 
name.

Rachel.

wykys, 2021-10-30:
I am also pleased with this initiative. Please enter a name Jan Vykydal.

esden, 2021-10-30:
The nord theme that I submitted is actually desigend by @jordiorlando I only did some conversion to add it here. I think the approval and credit should go to him. If it was up to me I would definitely be fine with getting the theme added. :)

pointhi, 2021-10-30:
This is a sneak peak how it looks right now:

[image]

DX-MON, 2021-10-31:
> This is a sneak peak how it looks right now:
>
> PCM example
> <https://user-images.githubusercontent.com/1652144/139558441-f0e15a5b-4268-4b48-bf94-dc5ffb22f6d6.png>

That's looking awesome!

We have a couple of layer fixes for 8-layer stuff in the Eagle inspired 
theme due which we'll try and land before this lands, but that should be 
all that needs doing for that theme.

Rachel.

pointhi, 2021-10-31:
@DX-MON maybe we can add a light and a dark Eagle theme. I suppose only the background color is different?

pointhi, 2021-10-31:
As KiCad introduced various new color tags over time, it would be nice if all themes we include in the package manager have all colors assigned. I did this for solarized in this commit: https://github.com/pointhi/kicad-color-schemes/commit/c6a545a9b3dd300a23ccded864762767056205a3

jordiorlando, 2021-11-09:
This looks great! And thanks for the credit @esden 😄

Shackmeister, 2022-01-04:
I have written a message to the person who provided the monokai theme on the forum.
Would be great to get in the content manager :)

ANWatson, 2022-01-06:
Personally, I'm happy for the monokai theme to be used. My name's Ash Watson. However, the original creator of monokai itself ought to be credited. I'm trying to find a contact for them now.

poeschlr, 2022-01-20:
Sorry for the late response. I kind of missed this (was somehow lost in my inbox)

I am ok with anyone using my work however they want.

jordiorlando, 2022-02-19:
@pointhi is there a way for you to change @0xdec to @jordiorlando in the plug-in manager? Or perhaps I should open an issue in GitLab?
[image]

Chrismettal, 2022-05-25:
Thanks for getting most themes into the addon manager already!

Any reason for Behave-dark to be missing from the manager though?